evert-arias/EasyButton #1: Multiple .onSequence for a unique button

laboiteare, 2018-09-23:
Hi ! First, thank you for this great library. It's make easier the button handling on my arduino projet.

I tried to use multiples sequences for a unique button but only the last declaration works. I know it's hard to implement but do you think you can could handle more than one event for a unique button ?

For example, with a 2000 timeout, you could run a callback if you press one time, and a different callback if you press 2 ou 3 times.

I think it would be great for doing some "settings menu".
But anyway, your library is already great :)

evert-arias, 2018-10-29:
@laboiteare Hi. Thank you very much for your feedback. I'm very sorry for the delay. Undoubtedly you have indicated a good point, I will be looking at the issue very soon to implement that recommended functionality. Again, thank you very much for your comment . 

Best regards

finchamp, 2019-02-20:
@evert-arias -- I would be interested in this as well. I love the library so far, it seems to be the best one out there. I am working on an intuitive light dimmer switch using a capacitive touch panel, and I could "trick" the controller into recognizing several tap-gestures, it would be a little simpler to just institute multiple supported .onSequence().

I don't see a way to put a bounty on this, but I would donate a little to save myself the time. :)

dakky, 2019-06-07:
same would be nice for `.onPressedFor`: at the moment the callback with the lowest time gets fired

Petros144, 2019-07-10:
Same Here, work for a automotive project and have only a single button to work with and want to implement 4 functions (press, long press, press 2x , press 3x). Would also donate a few bucks to this nice Lib!!!

piotrpog72, 2020-01-02:
Hello.. is there something happens since February? I need this functionality very much in my present project.

evert-arias, 2020-01-03:
Hi @piotrpog72 
You will find this feature in version 2.0 that will be available soon.
Regards

finchamp, 2020-01-10:
Super!

evert-arias, 2020-03-23:
This feature has been added to EasyButton [v1.2.0](https://github.com/evert-arias/EasyButton/releases/tag/v1.2.0)

Also check https://easybtn.earias.me for the official documentation, examples and more.

evert-arias, 2020-03-23:
This issue has been resolved and therefore I will close it. Feel free to open a new problem if you need to.

closed 2020-03-23 by evert-arias

zener79, 2020-10-05:
Hi guys,
I'm re-opening this just to better understand the multiple onSequence behaviour.

Running [this example](https://github.com/evert-arias/EasyButton/commit/9b27abf162ab630a3be83ae562cc7b58d3e71ebd) if I quickly push the button 3 times I receive the following output.

`Double click`
`Other sequence`

Basically both the callbacks get triggered. Is this a desired behaviour?