evert-arias/EasyButton #12: Error using library

sunishnet, 2019-07-03:
I get the following error message 

EasyButtonTouch.cpp:21:23: error: 'touchRead' was not declared in this scope
  return touchRead(_pin) < _touch_threshold;

while trying to verify the code in Arduino/VSCode IDE. This happens whether the board is Arduino Uno or ESP8266

evert-arias, 2019-07-03:
Capacitive touch GPIOs are only available in ESP32. I'll highlight it in the library documentation.

sunishnet, 2019-07-04:
That's understandable, but why is the touch code getting invoked on a sample with normal switch on ESP8266 and Uno ?

evert-arias, 2019-07-04:
You're right, the touch code should only be invoked for ESP32. 
I have already applied the solution recommended by @gutierrezps at https://github.com/evert-arias/EasyButton/pull/11

A new version has been released https://github.com/evert-arias/EasyButton/releases/tag/v1.0.2

Thanks

sunishnet, 2019-07-05:
I checked the new version and it compiles and works fine.
Thanks

closed 2019-07-05 by sunishnet