evert-arias/EasyButton #14: Long button press issues

malaki86, 2019-07-04:
I'm trying out this library, but having issues, mainly with the long press function.

When the code starts, it almost instantly detects a long press. I have tried adding a delay (2 seconds) at startup, prior to initializing the button, but it still happens.

I also would like to have the button as dual-function (short & long press). When using both functions, both short & long press triggers the long press function. The code below is the current version, which only triggers the long press function.

`#include <EasyButton.h>

// pin assignments
const byte
  btnHeatPin(2); //D2

#define LongPress 2000

EasyButton btnHeat(btnHeatPin);

void btnHeatShortPress() {
  
  Serial.println("Button Heat short press");
}

void btnHeatLongPress() {
  Serial.println("Button Heat long press");
}

void setup() {
  Serial.begin(9600);
  Serial.println("Startup");
  btnHeat.begin();
  btnHeat.onPressedFor(LongPress, btnHeatLongPress);
  btnHeat.onPressed(btnHeatShortPress);
}

void loop() {
  btnHeat.read();
}`

malaki86, 2019-07-04:
Ok - I feel like a complete idiot now. I had a wire hooked up wrong. The code I posted above works 100%, having both short & long press functions.

evert-arias, 2019-07-04:
Hi @malaki86 
Don't worry, that happens all the time :smile:
Feel free to post any further issue that you may have or suggestions that may improve this library.

Regards

evert-arias, 2019-07-06:
I'm closing this issue since this is not a problem with the library

closed 2019-07-06 by evert-arias