evert-arias/EasyButton #21: Compile errors in MultipleButtons.ino

vazquezjm, 2019-08-18:
HI @evert-arias !

I'm getting the following error when compiling the example `MultipleButtons.ino`

```
\Arduino\libraries\EasyButton\src\EasyButton.cpp: In member function 'void EasyButton::enableInterrupt(EasyButton::callback_t)':

\Arduino\libraries\EasyButton\src\EasyButton.cpp:174:63: error: cannot convert 'EasyButton::callback_t {aka std::function<void()>}' to 'void (*)()' for argument '2' to 'void attachInterrupt(uint8_t, void (*)(), int)'

  attachInterrupt(digitalPinToInterrupt(_pin), callback, CHANGE);

                                                               ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
```

This happens both in Arduino IDE v1.8.9 and PlatformIO (VSC)

elC0mpa, 2019-08-18:
@vazquezjm thanks for inform us about this issue. We are working on it, it will be fixed in the Next release, wich will be ready soon

evert-arias, 2019-08-18:
Hi @vazquezjm 
This issue has been fixed https://github.com/evert-arias/EasyButton/commit/fed0ad89c017bb470ac818d4f93ec8c3a5f84b8a and a new release it's now available to [download](https://github.com/evert-arias/EasyButton/releases/tag/v1.1.1).
https://github.com/evert-arias/EasyButton/releases/tag/v1.1.1

We appreciate you feedback

vazquezjm, 2019-08-18:
Wow, that was super fast!

Quick note: I added the following line to my project `platformio.ini` in order to start using 1.1.1 as described [here](https://docs.platformio.org/en/latest/userguide/lib/cmd_install.html#description)

```lib_deps = EasyButton@1.1.1```

Although it looks like it is trying to download 1.1.1, I get this message:

[image]

Is that a PlatformIO issue?

evert-arias, 2019-08-18:
@vazquezjm 
PlatformIO takes a few hours even a day to update its libraries registry. 
I suggest you wait a couple of hours until the new version is available on PlatformIO. 
https://platformio.org/lib/show/5741/EasyButton

evert-arias, 2019-08-21:
This issue has been solved

closed 2019-08-21 by evert-arias