evert-arias/EasyButton #23: Using EasyButton with Low-Power

vazquezjm, 2019-08-31:
Hi there!

This is not an issue per-se, but I need your help clarifying this.

Currently I'm working on an irrigation project with Arduino. This setup is stand-alone, no connections to external apps or servers. This is challenging since I have buttons for:

* button 1: enter programming mode while pressed
* button 2: while on programming mode, switch from max moisture to min (swap values shown on LED display)
* button 3: increment value
* button 4: decrement value

Since I'm going to use a low power library to manage sleep modes and stuff, I also need a way to manually awake the device from sleep. I thought about using another button but that would be too much, so I came across PressedForDuration example from your library. I was wondering if I could use button 1 for waking up the Arduino (one press) and enter programming mode if pressed for 2-3 secs? 

I order to wake it up, I need to use something like this:
https://github.com/rocketscream/Low-Power/blob/master/Examples/powerDownWakeExternalInterrupt/powerDownWakeExternalInterrupt.ino

My questions is, could I make both your library and Low Power work together? My main concern is having a button doing two things and working with different level (HIGH to enter programming mode, LOW to generate an interrupt and wake the board from sleep)

Thanks in advance with your help!

Jose

evert-arias, 2019-09-09:
Hi @vazquezjm
@elC0mpa it's going to help you with that. He should be around soon.

elC0mpa, 2019-09-11:
Hi @vazquezjm, basically you want to know if microcontroller could be awaked By using external interrupts. As you should know, powerdown library has a powerdownWakeExternalInterrupt example. So, microcontroller could be awaked By using external interrupts (buttons in this case). The problem is that when microcontroller is in powerdown state, timers are disabled, and timers are indirectly used By easybutton library, so this could be a problem. I think you should try it and give us a feedback for us to learn from you. Please let me know if you are pleased with this answer.

vazquezjm, 2019-09-11:
Thanks for the reply @elC0mpa 👍 

I was able to make it work this way, with 5 buttons:

[image]

```c++
#include "LowPower.h"

// Use pin 2 as wake up pin
const int wakeUpPin = 2;
int a = 0;
bool interrupt = false;

void wakeUp()
{
  // Just a handler for the pin interrupt.
  interrupt = true;

}

void setup()
{
  Serial.begin(9600);
  // Configure wake up pin as input.
  // This will consumes few uA of current.
  pinMode(wakeUpPin, INPUT_PULLUP);
  delay(50);
}

void loop()
{
  // Allow wake up pin to trigger interrupt on low.
  attachInterrupt(0, wakeUp, LOW);

  // Enter power down state with ADC and BOD module disabled.
  // Wake up when wake up pin is low.
  Serial.println("Sleep for 8s....");
  delay(200);
  LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);

  // Disable external pin interrupt on wake up pin.
  detachInterrupt(0);

  // Do something here
  if (interrupt)
  {
    Serial.println("Interrupt wake up");
    interrupt = false;
  } else {
    Serial.println("After 8s wake up");
  }
  delay(500);
  Serial.println("Do something....");
  for (a = 1; a <= 10; a++) {
    delay(200);
    Serial.print(".");
  }
  Serial.println("");
}
```

So now it goes to sleep for 8s and I can wake it up using a push-button. 

Now, my goal is to use `PressedForDuration` in order to have only 4 buttons like this:

- button 1: 
  - enter programming mode when pressed for 3 seconds
  - exit programming mode when pressed for 3 seconds
  - while on programming mode, switch from max moisture to min settings when pressed once
- button 2: increment value
- button 3: decrement value
- button 4: wake from sleep

Do you think it is feasible? 

I have a lot of work ahead 😅

evert-arias, 2020-03-18:
Hi @vazquezjm 
I am sorry I took so long to respond, I have been busy on other projects. I hope you have managed to make it work the way you wanted. 👍

evert-arias, 2020-03-18:
I consider that this issue has been resolved and therefore I will be closing it. Feel free to open a new issue if you need to.

closed 2020-03-18 by evert-arias