evert-arias/EasyButton #24: Can I make an Easy Button as a member of a class?

rgb-empire, 2019-09-08:


evert-arias, 2019-09-09:
Hi @rgb-empire 
Yes, you can make it a member of a class. I will add an example on that.

rgb-empire, 2019-09-09:
Awesome! I was trying but couldn't get the event function to work when it also belonged to that same class.

rgb-empire, 2019-09-10:
This is what I'm trying to do, any idea why it's not working?

void Physical_Input::best_button_on_pressed()
{
//do something
}

Physical_Input::Physical_Input()
	:enabled(true),
	best_button(EasyButton(_Best_Button_Pin)),
	rotary_button(EasyButton(_Rotary_Button_Pin)),
	encoder(Encoder(_Rotary_Pin_A,_Rotary_Pin_B))
{
        best_button.onPressed(Physical_Input::best_button_on_pressed);
}

rgb-empire, 2019-09-14:
@evert-arias 

So I think I inadvertantly figured this out while doing another class in my program.

I ended up needing to define a static member function to wrap the callback function in so that it could be used for the callback. I will update this post with code when I confirm that the same thing works for EasyButton!

evert-arias, 2020-03-18:
I am sorry I took so long to respond, I have been busy on other projects. I hope you have managed to make it work the way you wanted.

evert-arias, 2020-03-18:
I consider that this issue has been resolved and therefore I will be closing it. Feel free to open a new issue if you need to.

closed 2020-03-18 by evert-arias