evert-arias/EasyButton #25: Does onPressed() have higher priority over onPressedFor()?

thkfighter, 2019-09-15:
` 
//...
void onButton3Pressed(){
//...
}
void button3PressedFiveSeconds(){
//...
}
void button3ISR()
{
  button3.read(INTERRUPT);
}

void setup()
{
//...
button3.begin();
button3.onPressedFor(5000, button3PressedFiveSeconds);
button3.onPressed(onButton3Pressed);
//...
if (button3.supportsInterrupt())
  button3.enableInterrupt(button3ISR);
}
void loop()
{
}
`
I pressed button3 for more than 5 seconds and then released it, onPressed() was called instead of onPressedFor().

elC0mpa, 2019-09-15:
Hello @thkfighter, are you sure that this es exactly what is happening? I ask you because evert and i tested this feature before including it to the library and everything was fine

thkfighter, 2019-09-15:
> Hello @thkfighter, are you sure that this es exactly what is happening? I ask you because @evert-arias and i tested this feature before including it to the library and everything was fine. In case this is what is really happening, I could recommend you to increase the debounce time of the button and try it again. Please give us your feedback.

I'm afraid so.
According to your response, I changed the debounce time from default 35ms to 200ms. It did not work either.
`EasyButton button3(buttonPins[2], 200, true, true);`
The board is [ESP-12S](https://wiki.ai-thinker.com/esp8266) from Ai-Thinker, with three buttons connected on GPIO 4, 5 and 16 respectively.

evert-arias, 2019-09-15:
Hi @thkfighter 
I'm going to run some tests on what you have pointed out and get back to you later. 
Regards

evert-arias, 2019-09-15:
@elC0mpa @thkfighter 
I was able to replay the misbehavior of the **onPressedFor** feature when using interrupts. It's indeed not working properly, as stated by @thkfighter. I'm currently working on a solution to that issue.

elC0mpa, 2019-09-15:
@thkfighter there is a mistake in your code. If you want to use the onPressedFor functionality through external interrupts, you should call the update method in the loop function

thkfighter, 2019-09-16:
> @thkfighter there is a mistake in your code. If you want to use the onPressedFor functionality through external interrupts, you should call the update method in the loop function as shown here: https://github.com/evert-arias/EasyButton/blob/master/examples/InterruptsOnPressedFor/InterruptsOnPressedFor.ino
> Please fix this in your code and let us know if everything is working as expected

@elC0mpa @evert-arias Thanks a lot. It works.

closed 2019-09-16 by evert-arias