evert-arias/EasyButton #26: Does the library EasyButton work well for 365x24h continuous application?

thkfighter, 2019-09-16:
Is there any timer variable that will overflow when the application runs for a long time?

elC0mpa, 2019-09-16:
Why are you asking this @thkfighter? Inside the library we only use the millis() function wich use the timer0 or timer1, I don't remember exactly wich of these is. But the value returned by this function overflows after more than a month.

thkfighter, 2019-09-17:
> Why are you asking this @thkfighter? Inside the library we only use the millis() function wich use the timer0 or timer1, I don't remember exactly wich of these is. But the value returned by this function overflows after more than a month. If you want to improve your skills about timing events in microcontrollers, check this link: https://github.com/elC0mpa/Arduino-Tutorials/tree/master/Timing%20Events

thkfighter, 2019-09-17:
> Why are you asking this @thkfighter? Inside the library we only use the millis() function wich use the timer0 or timer1, I don't remember exactly wich of these is. But the value returned by this [...]

I wonder if the button will fail when the timer variable overflows.

closed 2019-11-05 by evert-arias