evert-arias/EasyButton #27: #define INTERRUPT 0

404-baitnotfound, 2019-09-21:
please remove the line "#define INTERRUPT 0"
it interferes with other Liberiers , and does not serve any purpose.

elC0mpa, 2019-09-21:
Hello @404-baitnotfound, thanks for giving us this advice. We had already realized about it, but this issue will be fixed in version two due to compatibility problems. I hope you could solve your problems right now By modifying the library locally in your pc.

closed 2019-09-25 by 404-baitnotfound

404-baitnotfound, 2019-09-25:
Since the PressedFor function does not give reliable results I stopped using the lib.
I wrote myself an button handling algorithm that can detect 
- short presses on released interrupt
- medium / long on pressed timer interrupt  (user does not have to release the button)
it works really well, and if someone is interested to make an general lib out of it I will share my code?