evert-arias/EasyButton #30: Callback for button release?

mrdc, 2019-11-12:
Hello,

Would be nice to add a callback for button release.

Thanks!

elC0mpa, 2019-11-12:
There is a callback for button release, is onPressed event, maybe what you mean is that it would be nice to have a onReleasedFor event. Thanks for your feedback

mrdc, 2019-11-12:
Yes, just for release event. I need to catch once when there is HIGH on pin, do some work and catch once LOW on pin. What I have now is that my Serial is flooded if I use wasPressed and wasReleased.

source-creator, 2020-02-21:
Greetings @elC0mpa 
The callback is not called for me when the button is released, only when it is pressed.
Perhaps I am missing something?

evert-arias, 2020-03-18:
Hi @source-creator 
Please provide your code to see if there is any issue and we'll try to reproduce it in our side.

gogreenpower, 2021-03-22:
Did anyone get a OnRelease Callback working?

eppfel, 2021-05-03:
I setup a check for `btn.isReleased()` after `btn.read()` once `pressedFor(X)`was triggered.

```
btn.read();
if (waspressedforsometime && btn.isReleased())
{
  onRelease();
}
```