evert-arias/EasyButton #31: imput Variable instead of imput Pin

Petros144, 2019-12-29:
Hi Evert Arias,

Thank you for your nice Lib! 

I want to ask if its possible to use the library with a normal Bool Variable 
instead of a Hardware Pin? or what kind of modifications are need to be make this work?

like instead of  "EasyButton button(BUTTON_PIN);"

for example "EasyButton button(BOOL_VARIABLE);"

evert-arias, 2020-01-03:
Hi @Petros144 
The first argument is a **uint8_t** data type. That's the parameter to set the pin where the button is connected. I don't understand why would you need to pass a boolean variable as the pin number. Could you elaborate more about this.

Petros144, 2020-01-03:
I work with CAN Bus data and all imput data like buttons arrive as  8 byte Hex values, I have already managed to get every button a specific variable to work with (like High / Low State). And Your library is perfect for my needs except that im not useing any Hardware pin on my Board.

So insted i want to work with my Button variable from the CAN Bus.

I hope you can understand what i want to achieve with my toughts.

evert-arias, 2020-01-06:
Hi @Petros144 
I understand what you are trying to achieve. Unfortunately, the EasyButton library is designed to work only with hardware buttons connected to the MCU's GPIO pins. However, you have pointed a good feature that could be added to the library: the possibility of using variables that act as virtual buttons or button signals that come from other sources instead of the local GPIO's of the MCU.
I cannot guarantee that this will be added any time soon, but I will do my best to include this feature in the next release.

Petros144, 2020-01-06:
Thank you very much! - Wil put some bucks in the Donations Jar!

evert-arias, 2020-04-11:
Hi @Petros144 
We just released [EasyButton v2.0](https://github.com/evert-arias/EasyButton/releases/tag/v2.0.0). This version has support for virtual buttons allowing you to use a variable instead of an actual 

EasyButton now has its own website where you can find the library documentation, examples and other useful information. https://easybtn.earias.me

We welcome any feedback from you. Thanks in advance

evert-arias, 2020-04-15:
I consider this issue to be solved and therefore I will close it. For any issue related to virtual buttons support in EasyButton v2.0 please open a new issue.

closed 2020-04-15 by evert-arias