evert-arias/EasyButton #32: ESP32 compilation issues

malaki86, 2020-03-01:
I'm trying to implement EasyButton on an ESP32 project, but I'm receiving errors during compiling.

Code:
`#include <EasyButton.h>

// pin assignments
const byte
  btnHeatPin(14);

#define LongPress 2000

EasyButton btnHeat(btnHeatPin);

void btnHeatShortPress() {
  
  Serial.println("Button Heat short press");
}

void btnHeatLongPress() {
  Serial.println("Button Heat long press");
}

void setup() {
  Serial.begin(9600);
  Serial.println("Startup");
  btnHeat.begin();
  btnHeat.onPressedFor(LongPress, btnHeatLongPress);
  btnHeat.onPressed(btnHeatShortPress);
}

void loop() {
  btnHeat.read();
}`

Error message:
`Arduino: 1.8.10 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp: In member function 'void EasyButton::onPressed(EasyButton::callback_t)':

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp:21:2: error: 'mPressedCallback' was not declared in this scope

  mPressedCallback = callback;

  ^

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp: In member function 'void EasyButton::onPressedFor(uint32_t, EasyButton::callback_t)':

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp:26:2: error: 'mPressedForCallback' was not declared in this scope

  mPressedForCallback = callback;

  ^

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp: In member function 'void EasyButton::onSequence(uint8_t, uint32_t, EasyButton::callback_t)':

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp:33:2: error: 'mPressedSequenceCallback' was not declared in this scope

  mPressedSequenceCallback = callback;

  ^

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp: At global scope:

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp:66:6: error: prototype for 'bool EasyButton::read()' does not match any in class 'EasyButton'

 bool EasyButton::read() {

      ^

In file included from C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton-2019-09-01_06-58-44-272.cpp:8:0:

C:\Users\Owner\Documents\Arduino\libraries\EasyButton\src\EasyButton.h:38:7: error: candidate is: bool EasyButton::read(int)

  bool read(int read_type = POLL);           // Returns the current debounced button state, true for pressed, false for released.

       ^

Multiple libraries were found for "EasyButton.h"
 Used: C:\Users\Owner\Documents\Arduino\libraries\EasyButton
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

malaki86, 2020-03-01:
As a note, I was trying to use version 1.1.1 when I received the error. I uninstalled and installed version 1.0.2 and it works perfect.

evert-arias, 2020-03-18:
Hi @malaki86 
I have tried to reproduce this error using your source code and **EasyButton v1.1.1** and it works fine as expected. It seems that the problem in your case is due to the EasyButton library not being installed correctly in your environment. 

Please refer to the [installation docs](https://easybtn.earias.me/docs/installation).