evert-arias/EasyButton #33: MCP 23017 Additional Inputs

mdleisure, 2020-03-05:
Thank you for this library it has made my introduction to ESP and Arduino a lot easier.

I have found myself in need of more inputs, I was surprised at how many inputs cant be used on the ESP boards even on the larger version.

Hence I am now looking into the MCP23017 to get more and allow futher development for both inputs and outputs. 

I have tried to read through and see if there is anyone using easy button on these exspansion boards and may be jumping the gun as my board has not arrived yet but can easy button define inputs from the 23017 easily.

I am going to have a play around and probably try using the adafruit library along with the wire library to see if its possible to mix and match the #define. Not sure if thats the way or just wake the bus and try to define the buttons manually before mapping them to easy button.

I am using the single press and pressed for part of easy button for the reading of the button presses and really dont want to lose that.

Thanks for any help

evert-arias, 2020-03-18:
Hi @mdleisure 
I appreciate your feedback.
It will definitely be interesting to make EasyButton capable of using the additional ports provided by chips like the **MCP23017**. That is a feature that would be very important to have in future versions of the library, therefore I will discuss the possibility.

In the meantime, feel free to share any ideas or the results of your tests. Any comments in this regard will be appreciated.

evert-arias, 2020-04-11:
Hi @mdleisure 
We just released [EasyButton v2.0](https://github.com/evert-arias/EasyButton/releases/tag/v2.0.0). This version has support for virtual buttons allowing you to use a port expander like the **MCP23017**. We will include an example using chips like the **MCP23017** in the coming days.

https://github.com/evert-arias/EasyButton/releases/tag/v2.0.0

mdleisure, 2020-04-14:
Hi,

Thankyou for the notification, Great news on the V2 release and the
expansion ports use.

I will change my project around and use all 6 buttons on the expansion port
to test and report any issues I may come across.

Again thank you for supporting and developing your library.

evert-arias, 2020-04-15:
I consider this issue to be solved and therefore I will close it. For any issue related to virtual buttons support in EasyButton v2.0 please open a new issue.

closed 2020-04-15 by evert-arias