evert-arias/EasyButton #39: onPressedForDuration not waiting for set duration?

vazquezjm, 2020-04-03:
Hi there.

I wired my test Arduino Pro Mini as follows:
[image]

When I run the `onPressedForDuration()` example (running on EasyButton v1.2.0)

```
#include <EasyButton.h>

// Arduino pin number where the button is connected.
#define BUTTON_PIN 9

// Duration.
int duration = 3000;

// Button.
EasyButton button(BUTTON_PIN);

// Callback.
void onPressedForDuration() {
  Serial.println("Button has been pressed for the given duration!");
}

void setup() {
  Serial.begin(9600);
 
  // Initialize the button.
  button.begin();

  // Attach callback.
  button.onPressedFor(duration, onPressedForDuration);
}

void loop() {
  // Continuously update the button state.
  button.read();
}
```

As soon as I press the button, I get "Button has been pressed for the given duration!", although I set duration to 3000. 

Any suggestion?

Thanks!

evert-arias, 2020-04-03:
Hi @vazquezjm I'll take a look at this and get back to you later on today.

elC0mpa, 2020-04-03:
Hi @vazquezjm, I am pretty sure that the problem is related to your hardware. You are connecting your button to vcc and you are using the library as if it  is connected to gnd. Please try the same connecting the button to gnd and let us know if it worked

vazquezjm, 2020-04-03:
> Please try the same connecting the button to gnd and let us know if it worked

[image]

It works now!

Is there a way I can use it connected to Vcc? Thx

evert-arias, 2020-04-03:
@vazquezjm 
Passing `false` as the third arguments on the EasyButton class constructor will invert the default button's logic as described in https://easybtn.earias.me/docs/fundamentals.

Please use the following example as reference:

```cpp
#define PWR_BTN_PIN 26
uint8 debounce = 35;
bool pullup = true;
bool invert = false;
EasyButton powerButton(PWR_BTN_PIN, debounce, pullup, invert);
```

vazquezjm, 2020-04-03:
Thank you @evert-arias, that did the trick!

closed 2020-04-03 by vazquezjm