evert-arias/EasyButton #40: Interrupt example not working with 1.2.1

octave, 2020-04-09:
With version 1.2.1 on Interrupt example:

`button.read(INTERRUPT);` generates: `error: 'INTERRUPT' was not declared in this scope`

evert-arias, 2020-04-09:
Hi @octave 
I will take a look at it and get back to you. I appreciate your feedback.

evert-arias, 2020-04-09:
@octave 
This is what is happening. In **v1.2.1** we fixed an issue with the `INTERRUPT` macro as it was already defined in the arduino environment. To fix such problem, we renamed `INTERRUPT` to `EASYBUTTON_READ_TYPE_INTERRUPT`. This update should have been released as a major version update since it was a breaking change. 

We will be fixing this error as well as update the corresponding examples in v2.0 soon. 
Meanwhile all you have to do is use `EASYBUTTON_READ_TYPE_INTERRUPT` instead of  `INTERRUPT` when calling the `read `method as follow:

```
void buttonISR()
{
 read() function
  button.read(EASYBUTTON_READ_TYPE_INTERRUPT);
}
```
I am very sorry for this inconvenience and I appreciate your feedback.

evert-arias, 2020-04-11:
Hi @octave
In version 2.0, that has been released today, there is no need to pass any arguments when calling the `button.read()` function. There more features included in this release, feel free to give it a try. 

https://github.com/evert-arias/EasyButton/releases/tag/v2.0.0

MattGarfield, 2020-04-23:
@evert-arias reminder to update the website documentation to reflect this change. I just updated to v2.0 and was spinning my wheels for a few hours until I found this thread.

https://easybtn.earias.me/docs/fundamentals#external-interrupts

MattGarfield, 2020-04-23:
The [Interrupts.ino example](https://github.com/evert-arias/EasyButton/blob/24d68cdfdd4081a1b3288295051d600e2ffb8eeb/examples/Interrupts/Interrupts.ino#L33) also still includes the comment to use the INTERRUPT parameter:

```
void buttonISR()
{
  /*
    When button is being used through external interrupts, 
    parameter INTERRUPT must be passed to read() function
   */
  button.read();
}
```

Awesome work on this super useful library. Thanks for all of your effort!!

evert-arias, 2020-05-16:
Hi @MattGarfield 
I will fix that as soon as possible. Thank you very much for your comment, I really appreciate your feedback.

pnaybour, 2021-09-04:
Has the documentation been updated yet?

goutamreddy, 2022-10-19:
Please update the documentation, 2022.10.19