evert-arias/EasyButton #44: Joystick / ThumbPad support, where button can have both low and high values

shafr, 2020-04-24:
Hi, I was woundering how to use your library to work with joystick buttons. They have basically x2 10K resistors, so values from analog read goes something like this:

* Left-Right/Top-Down: 

| Left | Middle   | Right |
|------|-----------|-------|
| 0    | 1850-1900 | 4096  |

* Joystick press:

| Pressed | Default  |
|---------|----------|
| 0       | 800-1300 |

I've tried both Pull_down and Pull_up for the corresponding pins, but values are the same.

Is there any way how I can define (ideally with interrupts) buttons that have both high and low values ? Or define max/min value ?

I'm using ESP32 DEV and [Joystick Like this](https://www.aliexpress.com/item/4000446127337.html)

Thank you.

evert-arias, 2020-04-29:
Hi @shafr 
At the moment the library does not support this type of control but you have definitely pointed out an interesting feature to be implemented.

We appreciate your feedback.

shafr, 2020-04-29:
@evert-arias thanks for comment!

I've found something similar that is already implemented [here](https://github.com/sudar/JoystickShield). It has calibration (3.3v and 5v) value detection and implementation itself. (Beerware license)

Maybe there can be a class that is `EasyJoystick` instead that would extend EasyButton and would have:
* triggered value and 
* default value range :) ?

evert-arias, 2020-05-16:
Hi @shafr It looks good to me 👍. It is a feature that I will take into account to work on in the future.

saikek, 2020-05-16:
@evert-arias Awesome. Thank you.