evert-arias/EasyButton #48: Unexpected behavior

ullix, 2020-10-19:
I am surprised by the results I am getting when defining multiple duration and sequence calls. Is this really the intended behavior of this lib?

This is the preogram, modified from your example:

        /*
         * MODIFIED from:
          Name:    MultipleSequence.ino
          Created:  03/23/2020 12:45:23 AM
          Author: José Gabriel Companioni Benítez (https://github.com/elC0mpa)
          Description: Example to demostrate how to work with multiple sequences 
        */
        
        #include <Arduino.h>
        #include <EasyButton.h>
        
        // Arduino pin where the button is connected to.
        #define BUTTON_PIN 0
        
        #define BAUDRATE 115200
        
        void onPressed()        {  Serial.println("Button pressed");}
        void onPressedFor2000() {  Serial.println("Button pressed for duration 2000"); }
        void onPressedFor5000() {  Serial.println("Button pressed for duration 5000"); }
        void DuoClick()         {  Serial.println("Double click"); }
        void TripleClick()      {  Serial.println("Triple click"); }
        void QuattroClick()     {  Serial.println("Quattro click");}
        
        // Instance of the button.
        EasyButton button(BUTTON_PIN);
        
        void setup()
        {
          // Initialize Serial for debuging purposes.
          Serial.begin(BAUDRATE);
        
          Serial.println();
          Serial.println(">>> EasyButton multiple onSequence example <<<");
        
          // Initialize the button.
          button.begin();
        
          button.onPressed(onPressed);
          button.onPressedFor(2000, onPressedFor2000);
          button.onPressedFor(5000, onPressedFor5000);
          button.onSequence(2, 1500, DuoClick);
          button.onSequence(3, 2500, TripleClick);
          button.onSequence(4, 3500, QuattroClick);
        }
        
        void loop()
        {
          // Continuously read the status of the button.
          button.read();
        }

The result is the following:

Pressing the button for 10 sec. After 5sec this is printed:

        Button pressed for duration 5000
I'd expect the print would occur only after 10 sec, but so far ok. Now I press the button for 3 sec:

        Button pressed
I'd expect the callback for duration 2000 to fire, but this actually never shows up.

Now I am pushing the button 4 times within 1 sec. This is the result:

        Button pressed
        Button pressed
        Double click
        Button pressed
        Triple click
        Button pressed
        Double click
        Quattro click

So, I am getting 4 single button presses, 2 double presses, 1 triple press and 1 quattro press, while all I was wanting was 1 quattro press.

Effectively I have only two options: one single press, plus  one long press. Everything else cannot be distinguished. Am I doing something wrong, or is that really intended?

elC0mpa, 2020-10-19:
Hi @ullix 
Nowadays this library **doesn't support multiple onPressedFor() events**, that is the reason why it only calls the `onPressedFor5000` function
Soon we will decide what to do about this and will let you know

JawadHyder, 2021-02-10:
I am observing the same behavior. All I want is to have at least 3 different actions on a single button. I can get 2 working fine (single press and long press) but can't add a double press sequence. Any progress on this issue?

ullix, 2021-02-11:
I solved this for me by moving to the [AceButton lib](https://www.arduino.cc/reference/en/libraries/acebutton/). I use now clicked, double clicked, and long pressed functions, and I can detect a button press during boot. Works well for me.

JawadHyder, 2021-02-12:
Thanks for your reply @ullix. I'll check out the AceButton lib. Although I was able to modify the EasyButton library to make it work by using esp8266 Ticker library to generate a one time callback after the maximum time duration has lapsed to only trigger the action associated with the maximum clicks but I'm not sure if this lib is supported by other boards so I'll probably stick with something tested across multiple boards.