evert-arias/EasyButton #50: nrf52 'NOT_AN_INTERRUPT' was not declared in this scope

packetloss69, 2020-11-28:
Hi there.
Issue: When using EasyButton > 1.02 compilation failure with Adafruit nrf52 BSP
Using https://github.com/adafruit/Adafruit_nRF52_Arduino BSP.
BSP versions affected 0.70 - 0.21.0 (All)

Works on all versions: Easybutton 1.0.2
Fails: 1.1.0 up to and including 2.0.1

Error produced:
`C:\Users\User\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp: In member function bool EasyButton::supportsInterrupt():
C:\Users\User\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp:139:41: error: NOT_AN_INTERRUPT was not declared in this scope
  139 |  return (digitalPinToInterrupt(_pin) != NOT_AN_INTERRUPT);`

`Error compiling for board Nordic nRF52840DK (PCA10056).`

in EasyButton.cpp:
`bool EasyButton::supportsInterrupt()
{
	return (digitalPinToInterrupt(_pin) != NOT_AN_INTERRUPT);
}`

references:
`C:\Users\User\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\cores\nRF5\Arduino.h`

contains

`#define digitalPinToInterrupt(P)   ( P )`

Any ideas?

CSC-Sendance, 2022-11-11:
A valid workaround for this that also avoids issues stemming from copying the whole library or trying to update it is including the macro via buildflag: `-DNOT_AN_INTERRUPT`=-1`

If you are using platformio, edit your platformio.ini 's build variant to include:

```
build_flags = 
	-DNOT_AN_INTERRUPT=-1 ;
```

jonfroehlich, 2022-12-14:
Just confirming that this is still an issue with Adafruit Feather nRF52840, Arduino IDE 2.0.3, and EasyButton Version 2.0.1.

```
c:\Users\jonf\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp: In member function 'bool EasyButton::supportsInterrupt()':
c:\Users\jonf\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp:99:41: error: 'NOT_AN_INTERRUPT' was not declared in this scope
   99 |  return (digitalPinToInterrupt(_pin) != NOT_AN_INTERRUPT);
```

I'm not using interrupts with buttons so I added `#define NOT_AN_INTERRUPT -1` to the top of EasyButton.h and that solved it.