evert-arias/EasyButton #54: Error with Digistump/attiny85 compiling

saddys, 2021-03-03:
Error with Digistump/attiny85 compiling. Any suggest? Thanks!

C:\Users\utente\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp: In member function 'bool EasyButton::supportsInterrupt()':
C:\Users\utente\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp:99:36: error: 'digitalPinToInterrupt' was not declared in this scope
  return (digitalPinToInterrupt(_pin) != NOT_AN_INTERRUPT);
                                    ^
C:\Users\utente\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp: In member function 'void EasyButton::enableInterrupt(EasyButtonBase::callback_t)':
C:\Users\utente\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp:104:44: error: 'digitalPinToInterrupt' was not declared in this scope
  attachInterrupt(digitalPinToInterrupt(_pin), callback, CHANGE);
                                            ^
C:\Users\utente\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp: In member function 'void EasyButton::disableInterrupt()':
C:\Users\utente\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp:110:44: error: 'digitalPinToInterrupt' was not declared in this scope
  detachInterrupt(digitalPinToInterrupt(_pin));
                                            ^
C:\Users\utente\Documents\Arduino\libraries\EasyButton\src\EasyButtonBase.cpp: In member function 'void EasyButtonBase::_checkPressedTime()':
C:\Users\utente\Documents\Arduino\libraries\EasyButton\src\EasyButtonBase.cpp:71:23: error: range-based 'for' loops are not allowed in C++98 mode
   for (Sequence seq : _sequences)
                       ^

TheLastGimbus, 2022-01-29:
This dosn't let me use this library at all on Attiny. Could someone take a look?

TheLastGimbus, 2022-01-30:
I added a hack in my fork. You can use it with platformio like:
```ini
[env:digispark-tiny]
platform = atmelavr
board = digispark-tiny
framework = arduino
lib_deps = 
	...
	git+git://github.com/TheLastGimbus/EasyButton#attiny85-hack
```