evert-arias/EasyButton #60: Multiple onPressedFor

FraOre, 2021-10-01:
Hi,
I would like to do something like this with the same button:

` 
button.onPressedFor(1000, showIP);
button.onPressedFor(3000, restartDevice);
`

so execute two different callback based on the pressing duration. It doesn't work for me. Just the last function is called.
Any idea?

Thank you so much,
Francesco

TylerTotally, 2022-01-27:
I am trying to do the same thing, and having the same problems.

evert-arias, 2022-03-12:
Hi @FraOre and @TylerTotally, I will take a look at this.

JFClaeys, 2022-03-31:
Basically, it is the last defined OnPressedFor  that is stamped in the EasyButtonBase class's members
So I believe that an array (or a chained list of events) should be set in place in order to hold the different OnPressedFor callbacks.
Of course, _checkPressedTime  should then loop that structure until found proper callback.
I'll see how to setup something decently behaving :)

fastbike, 2022-09-22:
This would be a useful addition to the library