evert-arias/EasyButton #69: Error example simple button with interrupt in ESP32.

oriaj3, 2022-05-30:
Hi, I am testing the library to include it in a larger project and replace my rudimentary library to use short press and long press on various buttons. But after compiling and uploading the example code to my ESP32 with VScode, I get this error. I use the single button example with interrupt and my ESP32 is made by doit, esp-wroom-32 devkit v1.

`#include <Arduino.h>
#include <EasyButton.h>

const uint8_t BOTON_ONOFF=23;

#define BAUDRATE 115200

// Instance of the button.
EasyButton button(23);

void IRAM_ATTR buttonPressed()
{
  Serial.println("Button pressed");
}

void IRAM_ATTR buttonISR()
{
  button.read();
}

void setup()
{

  // Initialize Serial for debuging purposes.
  Serial.begin(BAUDRATE);

  Serial.println(">>> EasyButton interrupts example <<<");

  // Initialize the button.
  button.begin();

  button.onPressed(buttonPressed);

  if (button.supportsInterrupt())
  {
    button.enableInterrupt(buttonISR);
    Serial.println("Button will be used through interrupts");
  }
}

void loop()
{
  // put your main code here, to run repeatedly:
  void(10);
}`

I get the following guru meditation error. 

`Button pressed
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008a3b6  PS      : 0x00060b35  A0      : 0x8008960a  A1      : 0x3ffbec9c  
A2      : 0x3ffb8a00  A3      : 0x3ffb8890  A4      : 0x00000004  A5      : 0x00060b23  
A6      : 0x00060b23  A7      : 0x00000001  A8      : 0x3ffb8890  A9      : 0x00000018  
A10     : 0x3ffb8890  A11     : 0x00000018  A12     : 0x3ffc1894  A13     : 0x00060b23  
A14     : 0x007bee88  A15     : 0x003fffff  SAR     : 0x00000017  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400860f9  LEND    : 0x40086109  LCOUNT  : 0xfffffffc  
Core  1 was running in ISR context:
EPC1    : 0x400d9a43  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x4008a3b3:0x3ffbec9c |<-CORRUPTED

Core  0 register dump:
PC      : 0x4008a537  PS      : 0x00060035  A0      : 0x80089233  A1      : 0x3ffbe7dc
A2      : 0x3ffbee88  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060023
A6      : 0x00060021  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0xffffffff
A10     : 0x3ffc16a0  A11     : 0x00000000  A12     : 0x3ffc169c  A13     : 0x00000007
A14     : 0x007bee88  A15     : 0x003fffff  SAR     : 0x0000001a  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4008a534:0x3ffbe7dc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
>>> EasyButton interrupts example <<<
Button will be used through interrupts`

Elemeants, 2022-06-09:
Hi!

I think your error is on this:

```cpp
void loop()
{
  // put your main code here, to run repeatedly:
  void(10);
}
```

The ESP32 internal Arduino implementation feeds the watchdog timer each time a loop function is called, so that exception is because the watchdog timer was triggered for a timeout.

So remove the `void(10)` code and try again.

sevenberyl, 2022-08-11:
Same error, even without void(10)：
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1).

ahmedmibrahim, 2022-08-18:
Same error here:
The error backtrace is logged below

Decoding stack results
0x4037c48f: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c line 182
0x4037b75f: vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h line 38
0x4037abaa: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1688
0x4200238f: uartWriteBuf at C:\Users\ahmed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32\esp32-hal-uart.c line 309
0x42001899: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\ahmed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32\HardwareSerial.cpp line 473
0x42001c8b: Print::write(char const*) at C:\Users\ahmed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/Print.h line 67
0x42001cb1: Print::println(char const*) at C:\Users\ahmed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32\Print.cpp line 195
0x42001375: buttonPressed() at G:\My Drive\Jobs\Upwork\14-Astrohaus---USA\Firmware\Other sketches\Interrupts/Interrupts.ino line 21
0x420014ce: EasyButton::read() at C:\Users\ahmed\Documents\Arduino\libraries\EasyButton\src\EasyButton.cpp line 61
0x42001392: buttonISR() at G:\My Drive\Jobs\Upwork\14-Astrohaus---USA\Firmware\Other sketches\Interrupts/Interrupts.ino line 35
0x4201eb15: __onPinInterrupt at C:\Users\ahmed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32\esp32-hal-gpio.c line 159
0x40375199: gpio_isr_loop at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/gpio.c line 416
0x403751be: gpio_intr_service at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/gpio.c line 433