evert-arias/EasyButton #79: New, fixed version v2.0.2 needs update to show up in Arduino Library Manager

webmonkey, 2023-05-20:
Please could you bump the version number in library.properties so that it can be loaded via the Arduino library manager.

offa, 2023-05-30:
This affects platformio user too. 2.0.1 is [listed as latest release](https://registry.platformio.org/libraries/evert-arias/EasyButton) there too.

offa, 2023-07-04:
ping @evert-arias

JamesCohen-awin, 2023-07-04:
@evert-arias 

I think that the fix in #80 isn't going to work because the Github tag is at the older release (with `library.properties` set to 2.0.1).

You'll either need to re-tag the current HEAD or create a new tag for 2.0.3 with an updated `library.properties`

webmonkey, 2023-07-04:
@evert-arias

I think that my fix in https://github.com/evert-arias/EasyButton/pull/80 isn't going to work because the Github tag is at the older release (with library.properties set to 2.0.1).

You'll either need to re-tag the current HEAD or create a new tag for 2.0.3 with an updated library.properties

evert-arias, 2023-07-04:
@webmonkey you are right, I'll create a new tag 2.0.3.

evert-arias, 2023-07-04:
Done!

evert-arias, 2023-07-04:
I will leave this issue open until we confirm that version 2.0.3 is available in the Arduino Library Manager.

webmonkey, 2023-07-04:
The new version is available via the library manager.

```
$ arduino-cli lib list --updatable
Name       Installed     Available     Location              Description

EasyButton 2.0.1         2.0.3         LIBRARY_LOCATION_USER Arduino library for debouncing moment...
```

PR #80 (2023-07-04): Bump version number in library.properties
fixes #79
commit 852351a1 (2023-07-04): Bump version number in library.properties

fixes #79
closed 2023-07-04 by webmonkey