evert-arias/EasyButton #83: Filter.h not found on v2.0.3. Failed to compile

m-gracia, 2023-07-08:
Hello.
First of all, thank you for your great job with this library.

I have been using this library on my project but now it throws a compilation error since v2.0.3:
```
In file included from .pio/libdeps/wemos_d1_mini32/EasyButton/src/EasyButtonTouch.cpp:9:
.pio/libdeps/wemos_d1_mini32/EasyButton/src/EasyButtonTouch.h:14:10: fatal error: Filter.h: No such file or directory
...
...
 #include <Filter.h>
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\wemos_d1_mini32\libe3f\EasyButton\EasyButtonTouch.cpp.o] Error 1
```

The same code compiles good on v2.0.1.
I am _almost_ sure it is not an error on my environment because I run a <full clean> before the compilation.
Also the MegunoLink dependency is resolved with MegunoLink@1.42.0

My env:
- PlatformIO
- Board: Wemos_D1_miniESP32
- Framework: Arduino

Best regards.

oriaj3, 2023-07-10:
Same error here.

robercy, 2023-07-14:
Same error here.

evert-arias, 2023-07-14:
I'll check it out

bohdanbirdie, 2023-08-15:
Can confirm this on `board = esp32doit-devkit-v1`

Speedy2k, 2023-11-22:
Any news on this ??
I got the same error on 2.0.3

libraries\EasyButton-2.0.3\src\EasyButtonTouch.h:14:10: fatal error: Filter.h: No such file or directory

mathieucarbou, 2023-11-23:
same problem, platformio, board esp32dev

mtojay, 2023-12-01:
hmmm no solution here? same problem. seems like its not down to user error if mutliple people have that problem?

mathieucarbou, 2023-12-01:
> hmmm no solution here? same problem. seems like its not down to user error if mutliple people have that problem?
> 
> also platformio in vscode and esp32dev board

Project seems dead. Look at this fork https://github.com/vdccc/EasyButton

This is the most recent one I've found, that cleans up the "mess" introduced in the latest commits in this project.

mtojay, 2023-12-01:
> > hmmm no solution here? same problem. seems like its not down to user error if mutliple people have that problem?
> > also platformio in vscode and esp32dev board
> 
> Project seems dead. Look at this fork [vdccc/EasyButton](https://github.com/vdccc/EasyButton)
> 
> This is the most recent one I've found, that cleans up the "mess" introduced in the latest commits in this project.
> 
> Otherwise, have a look at [rwmingis/InterruptButton](https://github.com/rwmingis/InterruptButton), which I find more reliable for the detection because it does not rely on the loop being called, both in mode sync and async.

amazing. thank you. i will look in to those two options!
much appreciated