evert-arias/EasyButton #88: solve the Filter.h not found problem

wanggaoteng, 2023-12-23:
I open this new issue just for convenient.
If you use Platformio, and install the EasyButton 2.0.3 library through the Platformio library installation, you can find that there is another library which named MegunoLink will also installed too. 
In the library folder of MegunoLink/src, there is the file Filter.h, so just copy it to the library foler of EasyButton\src, problem solved. 
[Filter.zip](https://github.com/evert-arias/EasyButton/files/13759056/Filter.zip)

madkiwinz, 2024-02-06:
@wanggaoteng thank you for the suggested fix of copying the Filter.h file to the EasyButton\src folder

jsolderitsch, 2024-07-29:
I am not working with Platformio but with the recent changes to the esp32 family within Arduino IDE, the code from MegunoLink now throws errors on compiles when I include EasyButton in my sketch. I removed the MegunoLink library and then the only error I get is that ```filter.h``` is missing. I followed @wanggaoteng's advice and grabbed the filter.h file from this issue and copied it to the src folder for EasyButton. Then the  compile error went away.

SwackSun, 2024-10-14:
Can also work: 

`#include "../../MegunoLink/src/Filter.h"`