evert-arias/EasyButton #96: EasyButton error with ESP32.

ledadrian, 2024-07-31:
I was trying to use the EasyButton library in wokwi.com, but it was failing due to this error (See images)

I then tried to compile the 'Interrupts' example on a ESP32 DEV board, but I got the same error. 

Swapping to another ESP32 board results in the same error. 

[image]
[image]

ledadrian, 2024-07-31:
This was with the "Interrupts" example with no changes. 
I also made sure to install the dependancy "MegunoLink"
[image]

[image]

DarwinWasWrong, 2024-08-01:
The issue is caused by the easybutton dependency on  MegunoLink which is only used for one file Filter.h in the touch section of the library.
If you could take the library dependency out and contact MegunoLink for permission to use the Filter.h  this would stop easy button from loading the entire library - which crashes on certain ESp32's

jomoengineer, 2024-10-17:
This is a bit dated, but for anyone else running into this issue, I found if I downgrade the MegunoLink library to version 1.41 the code compiles without issue. Something in 1.42 is causing the issue. 
Note: I am using a ESP32 HitLetgo ESP32-WROOM-32D.

jsolderitsch, 2024-11-11:
MegunoLink has released library version 1.43 which fixes the error. This version is available with the Arduino Library Manager as of November 11 2024.

elC0mpa, 2024-11-13:
s