evert-arias/EasyButton #99: Documentation domain expired

HeavyFalcon678, 2025-01-30:
When I go to the link to read the documentation, then there is a message saying that the domain is expired. Could you please either renew the domain, or to add the docs here? Much appreciated!

marmilicious, 2025-01-30:
In the mean time you might be able to find some of what you were looking for here:

https://web.archive.org/web/20231020203303/https://easybtn.earias.me/

evert-arias, 2025-03-03:
@HeavyFalcon678 @marmilicious Thank you for bringing this to my attention. The library documentation has been moved to a new location: [https://evert-arias.github.io/easybtn-docs](https://evert-arias.github.io/easybtn-docs)
The previous domain has expired and won't be renewed. Please use this new GitHub Pages URL for all documentation needs going forward.
It is planned for the library to have its own domain in the future, but for now, this GitHub Pages URL will serve as the official documentation site.

evert-arias, 2025-03-03:
I'm closing this issue as it has now been resolved. The documentation is available at [https://evert-arias.github.io/easybtn-docs ](https://evert-arias.github.io/easybtn-docs )and will remain there until further notice.

closed 2025-03-03 by evert-arias